UBC-Snowbots/RoverFlake #2: Rewrite Arm firmware + driver with a more sophisticated IDE

Roozki, 2023-08-18:
Re write arm firmware for Inverse kinematics and more robust communication
End Goal:
1. Be able to control the arm by setting target positions, velocities, or accelerations. (Position is most important)
2. Decide whether we want to move to CAN, or if serial communication is enough for our purposes.
3. Absolutely no drift on any axis
4. Have a way to reset teensy remotley
5. Have a heartbeat sequence - (this could just be the position feedback)
6. Nail down specific communication rates. Lets aim for an update rate of 20hz (newer robot arms get up to like 1000hz which is nuts, avg human reaction rate is about 4 to 5 hz)
7. have a fail safe if comms break, arm should stop all movements after a specified amount of time without input (something like 0.5 seconds, this would mean the driver missed 10 command messages in a row)
8. Write some test code for limits, and write a gui for Arm subteam. It would be usefull if the arm subteam could test repeatability, set in their limits manually.

Roozki, 2023-08-18:
Attempting with Zephyr RTOS and serial communication.

Roozki, 2023-09-16:
Currently have velocity control via joystick. Currently works, but needs more features and testing

jodhsinghnahal, 2023-09-23:
hi