UBC-Snowbots/RoverFlake #3: Create a Captain/Control/Operator Station

Roozki, 2023-08-18:
We need to manufacture a control station.
I'm thinking it will more or less replace our GUI, but through digital displays, physical buttons and whatnot.

Goal:
Have a crate/station that is easy/quick to set up, and easy to control. This is to make competition software setup easier. We should do this soon so we can practice with it and get comfortable with it

Requirements:
Easy to move by one or two people.
    - Under 15kg
    - Carry Handles
    - able to be dragged around and bumped
Dust proof when closed
Operators only need to connect power and a CAT-5 Ethernet cable
Robust, can survive waist-height drops (the more robust the better)
Physical buttons to initiate robot
Software "e stop"
digital displays for battery levels
digital display for a timer, to show task time remaining
Common Peripherals (Keyboard, mouse, video display)
We can add much, much more if we want to.

kimi0324, 2023-09-23:
hello Kimi wants to do this thanks

jodhsinghnahal, 2023-10-07:
interested