UBC-Snowbots/RoverFlake #7: CIRC2024 Communications Base / Network Top Level Issue

Roozki, 2023-09-09:
### Description
This master issue outlines the core developments and enhancements aimed at improving communications and networking for the UBC Rover in 2023. We intend to fix a GNSS base station to the communications base, automate parabolic antenna tracking toward the rover, and implement network speed and data transfer monitoring.

### Objectives
- Implement a fixed GNSS base station at the comms base
- Automate the parabolic antenna to point toward the rover
- Monitor Wi-Fi speed and data transfer rates

### Milestones
- Complete GNSS base station setup and initial calibration
- Prototype automatic parabolic antenna pointing
- Complete and test automated antenna pointing
- Implement and test Wi-Fi speed and data monitoring systems

### What We Are Working With
**Hardware**
GNSS Base Station: For high-accuracy rover positioning
Parabolic Antenna: For long-range communication with the rover
Servo/Stepper Motors: To enable antenna movement

**Software**
- GNSS Software: For base station configuration and data handling
- ROS or other control algorithms: For automated antenna pointing
- Custom Networking Tools: For Wi-Fi speed and data monitoring

### Resources
- [GNSS Base Station Documentation](https://your-link-here/)
- [Antenna Pointing Algorithms Paper or Tutorial](https://your-link-here/)
- [Wi-Fi and Network Monitoring Tools Documentation](https://your-link-here/)

Roozki, 2023-09-09:
Max - Communications Lead

        - $300 budget 
        - requirements
        - being able to read the gps from base, without connection
        - add radio,

deadfloppy, 2023-09-09:
Also, quantify and expand bandwidth on Wi-Fi to fit multiple camera streams