UBC-Snowbots/RoverFlake #34: Use Zephyr RTOS to create serial driver for Magnemometer sensor

Roozki, 2023-10-29:


stephenqiao1, 2023-11-17:
Was able to detect the imu sensor with the esp32 and zephyr rtos, but still trying to figure out why it isn't able to read and process the magnetometer data

Roozki, 2023-11-18:
nice work so far, your code looks good to me. what is it outputting?
also how well is the serial connection? your code looks much cleaner than my implementation lol

stephenqiao1, 2023-11-19:
Im not sure what you mean by how good the serial connection is but the output is like this right now:
[image]

Roozki, 2023-11-20:
yeah, i want to make some packages that test serial speed and whatnot but if you haven't noticed any problems with the serial connection (connects every time, no garbage values, sends at the rates you want it to, etc) then it should be ok. That output might be something messing up with changing data types, or the IMU is just faulty

stephenqiao1, 2023-11-24:
i used the samples they had on the zephyr site. Im trying to debug by printing log messages from the driver source files, but for some reason I cant print the LOG_DBG() messages